hardkernel/ODROID-GO #3: Linking fails with es-idf

lfuelling, 2018-09-22:
Hi,

I'm trying to get the minimal HelloWorld to build using esp-idf on the command line because I don't like the ArduinoIDE.

I got the compilation to work, but when make tries to link it fails with a lot of errors:

```
> make                                                                                                                                                                                                                                                                                                
Python requirements from /Users/lerk/esp/esp-idf/requirements.txt are satisfied.
CXX build/main/main.o
AR build/main/libmain.a
LD build/odroid-go-hello-world.elf
/Users/lerk/workspace/odroid-go-hello-world/build/main/libmain.a(main.o):(.literal._Z4loopv+0x8): undefined reference to `ILI9341::clearDisplay()'
/Users/lerk/workspace/odroid-go-hello-world/build/main/libmain.a(main.o):(.literal._Z4loopv+0xc): undefined reference to `ILI9341::setTextSize(unsigned char)'
/Users/lerk/workspace/odroid-go-hello-world/build/main/libmain.a(main.o):(.literal._Z4loopv+0x10): undefined reference to `ILI9341::setTextFont(unsigned char)'
/Users/lerk/workspace/odroid-go-hello-world/build/main/libmain.a(main.o): In function `loop()':
/Users/lerk/workspace/odroid-go-hello-world/main/main.cpp:10: undefined reference to `ILI9341::clearDisplay()'
/Users/lerk/workspace/odroid-go-hello-world/main/main.cpp:11: undefined reference to `ILI9341::setTextSize(unsigned char)'
/Users/lerk/workspace/odroid-go-hello-world/main/main.cpp:12: undefined reference to `ILI9341::setTextFont(unsigned char)'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o):(.literal._ZN7ILI9341D5Ev[ILI9341::~ILI9341()]+0x0): undefined reference to `vtable for ILI9341'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o):(.literal._ZN9ODROID_GOC5Ev[ODROID_GO::ODROID_GO()]+0x0): undefined reference to `Button::Button(unsigned char, unsigned char, unsigned int)'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o):(.literal._ZN9ODROID_GOC5Ev[ODROID_GO::ODROID_GO()]+0x4): undefined reference to `ILI9341::ILI9341(short, short)'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o):(.literal._ZN9ODROID_GOC5Ev[ODROID_GO::ODROID_GO()]+0x8): undefined reference to `SPEAKER::SPEAKER()'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o):(.literal._ZN9ODROID_GOC5Ev[ODROID_GO::ODROID_GO()]+0xc): undefined reference to `Battery::Battery()'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o):(.literal._ZN9ODROID_GO5beginEm+0x10): undefined reference to `SPEAKER::begin()'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o):(.literal._ZN9ODROID_GO5beginEm+0x14): undefined reference to `ILI9341::begin()'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o):(.literal._ZN9ODROID_GO5beginEm+0x18): undefined reference to `ILI9341::setRotation(unsigned char)'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o):(.literal._ZN9ODROID_GO5beginEm+0x1c): undefined reference to `ILI9341::fillScreen(unsigned int)'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o):(.literal._ZN9ODROID_GO5beginEm+0x20): undefined reference to `ILI9341::setCursor(short, short)'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o):(.literal._ZN9ODROID_GO5beginEm+0x24): undefined reference to `ILI9341::setTextColor(unsigned short)'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o):(.literal._ZN9ODROID_GO5beginEm+0x28): undefined reference to `ILI9341::setBrightness(unsigned char)'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o):(.literal._ZN9ODROID_GO5beginEm+0x2c): undefined reference to `Battery::begin()'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o): In function `ODROID_GO::ODROID_GO()':
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.h:21: undefined reference to `Button::Button(unsigned char, unsigned char, unsigned int)'
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.h:21: undefined reference to `Button::Button(unsigned char, unsigned char, unsigned int)'
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.h:21: undefined reference to `Button::Button(unsigned char, unsigned char, unsigned int)'
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.h:21: undefined reference to `Button::Button(unsigned char, unsigned char, unsigned int)'
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.h:21: undefined reference to `Button::Button(unsigned char, unsigned char, unsigned int)'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o):/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.h:21: more undefined references to `Button::Button(unsigned char, unsigned char, unsigned int)' follow
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o): In function `ODROID_GO::ODROID_GO()':
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.h:21: undefined reference to `ILI9341::ILI9341(short, short)'
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.h:21: undefined reference to `SPEAKER::SPEAKER()'
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.h:21: undefined reference to `Battery::Battery()'
/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go/libodroid-go.a(odroid_go.o): In function `ODROID_GO::begin(unsigned long)':
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.cpp:12: undefined reference to `SPEAKER::begin()'
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.cpp:21: undefined reference to `ILI9341::begin()'
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.cpp:22: undefined reference to `ILI9341::setRotation(unsigned char)'
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.cpp:23: undefined reference to `ILI9341::fillScreen(unsigned int)'
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.cpp:26: undefined reference to `ILI9341::setCursor(short, short)'
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.cpp:27: undefined reference to `ILI9341::setTextColor(unsigned short)'
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.cpp:28: undefined reference to `ILI9341::setTextSize(unsigned char)'
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.cpp:29: undefined reference to `ILI9341::setBrightness(unsigned char)'
/Users/lerk/workspace/odroid-go-hello-world/components/odroid-go/src/odroid_go.cpp:30: undefined reference to `Battery::begin()'
collect2: error: ld returned 1 exit status
make: *** [/Users/lerk/workspace/odroid-go-hello-world/build/odroid-go-hello-world.elf] Error 1
```

I wonder what causes this, because as said before the compilation works without any errors.

Of course I Googled the problem before creating this issue and the only thing I could find that remotely looks like my problem [is this one](https://github.com/queezythegreat/arduino-cmake/issues/22#issuecomment-3810244) which suggests creating a `CMakeLists.txt` which also didn't work because I'm not using cmake.

seungchar, 2018-09-27:
Hi,
The odroid_go is only a library for the Arduino framework.
We have not yet made a library for esp-idf.

But, you can use the [loboris/ESP32_TFT_library](https://github.com/loboris/ESP32_TFT_library).

lfuelling, 2018-09-27:
Okay, thanks :)

I think I'll build myself a lib using OtherCrashOverride's OSS code.

But does this mean that there will be a library for esp-idf at some point?

seungchar, 2018-09-27:
Actually, We don't have the plan to make the library yet.
If you make some simple libraries for esp-idf, we can make an esp-idf section with your libraries on our [wiki](https://wiki.odroid.com/odroid_go/odroid_go#tutorials) :)

Thank you.

closed 2018-09-27 by lfuelling