hardkernel/ODROID-GO #4: Prebuild contains .ino.bin

s-hemer, 2018-10-11:
Can you clarify why the preuild folder contains .ino.bin files instead of .fw files ? Just asking, I sure can build them myself but beginners may be confused.

joshua-yang, 2018-10-16:
Hi.

The prebuild folder only contains .ino.bin files since they designed for those who doesn't use on device launcher. :)

s-hemer, 2018-10-17:
Thanks for the reply. But then the description is misleading and, from my point of view, the "complicated" step is to build the .fw file. Some parameters are really not well documented at mkfw tool.

32teeth, 2019-06-13:
I use a bash script to do this for me. might be of use to you

## Folder Structure
> On you desktop

* ESP32
  * Arduino
    *    assets
    *    firmware
    *    raw
  * scripts
    *    arduino.sh (see below)

Navigate to (in this example) and give bash file privelage

```shell
cd ESP32/scripts
chmod 777 arduino.sh
./arduino.sh
```

<details><summary>Bash File</summary>
<p>

```shell
#!/bin/zsh

SRC_PATH=`pwd`

#---------------------------------
# EDIT SKECTH NAME
#---------------------------------
REPO=$(dirname $PWD)
MKFW_PATH="$REPO/Odroid/odroid-go-firmware/tools/mkfw" # <-- change to wherever you put mkfw
ARDUINO_PATH="$REPO/Arduino"
SKETCH="Buttons" # <-- change the name and leave out .ino.bin

#---------------------------------
# DON'T TOUCH START
#---------------------------------
REPO=$(dirname $PWD)

#---------------------------------
# FANCY STUFF
#---------------------------------
export nc='\e[0m' # No Color
export red='\e[1;31m'
export green='\e[1;32m'
export yellow='\e[1;33m'
export blue='\e[1;34m'
export magenta='\e[1;35m'
export cyan='\e[1;36m'
export white='\e[1;37m'
export today=`date +%A\ %B\ %d\ %Y\ %T`

export squid=🐙
export robot=🤖
export unicorn=🦄
export fox=🦊
export lion=🦁
export monkey=🐵
export chicken=🐔
export panda=🐼
export whale=🐳
export game=🕹️

echo "\n--------------------------------------------------------------------------"
echo "\n${panda}  ${green}Creating ${cyan}Arduino${nc} Firmware"
echo "\n--------------------------------------------------------------------------"

cd $SRC_PATH

echo "Building mkfw"
cd $MKFW_PATH
make 

cd $SRC_PATH
ffmpeg -i $ARDUINO_PATH/assets/$SKETCH.png -f rawvideo -pix_fmt rgb565 $ARDUINO_PATH/raw/$SKETCH.raw -y

$MKFW_PATH/mkfw $SKETCH $ARDUINO_PATH/raw/$SKETCH.raw \
0 16 1048576 $SKETCH $ARDUINO_PATH/bin/$SKETCH.ino.bin

mv firmware.fw $ARDUINO_PATH/firmware/$SKETCH.fw

clear
echo "\n--------------------------------------------------------------------------"
echo "\n${unicorn}  ${green}Copmleted ${cyan}Gaboze Express${nc}"
echo "\n--------------------------------------------------------------------------"
exit
#---------------------------------
# DON'T TOUCH END
#---------------------------------
```
</p>
</details>

Be sure to check out [Gaboze Express](https://github.com/gaboze-express/GabozeExpress/tree/Software) for more information

hagen-git, 2021-02-22:
> Can you clarify why the preuild folder contains .ino.bin files instead of .fw files ? Just asking, I sure can build them myself but beginners may be confused.

For now, I added an explanation in the README.md file, hope that improves things a bit.