hardkernel/ODROID-GO #6: Hello world compiling error.

sok63, 2018-10-25:
I update my Arduino to latest ( when follow instructions to install support for ESP32 boards), and do all that was in instruction.

When try to compile Hello World it fall with next errors:

```
C:\Users\Сок\Documents\Arduino\libraries\ODROID-GO\src\sensors\RTClib.cpp: In constructor 'DateTime::DateTime(const char*, const char*)':

C:\Users\Сок\Documents\Arduino\libraries\ODROID-GO\src\sensors\RTClib.cpp:156:64: error: operation on '((DateTime*)this)->DateTime::m' may be undefined [-Werror=sequence-point]

   case 'J': m = date[1] == 'a' ? 1 : m = date[2] == 'n' ? 6 : 7; break;

                                                                ^

C:\Users\Сок\Documents\Arduino\libraries\ODROID-GO\src\sensors\RTClib.cpp: In member function 'uint8_t DS3231::begin()':

C:\Users\Сок\Documents\Arduino\libraries\ODROID-GO\src\sensors\RTClib.cpp:419:1: error: no return statement in function returning non-void [-Werror=return-type]

 }
```

Other GO sketches give other errors like:
```

C:\Users\Сок\Documents\Arduino\libraries\ODROID-GO\src\sensors\tca8418.cpp: In member function 'bool KEYS::configureKeys(uint8_t, uint16_t, uint8_t)':

C:\Users\Сок\Documents\Arduino\libraries\ODROID-GO\src\sensors\tca8418.cpp:81:1: error: no return statement in function returning non-void [-Werror=return-type]

 }
```

From all sketches only "LED\LED PWM" works (compile and upload) fine.

joshua-yang, 2018-10-26:
Hello.

I've just updated my Arduino program to the latest 1.8.7, but it seems occurs no error for me. :0
Are there any changes to your ODROID-GO library directory?

sok63, 2018-10-26:
Just try to install to clear laptop - and all works. After comparison the differences find the reason:
In main computer in Arduino-File-Prefrences-Compiler Warnings setted "All", but with clear install it setted to defaults. After I change to "Defaults" - all started to compile. I thought that this parameter simply prints more warnings, but as it turned out, it forces to compile with more strict rules.

per1234, 2018-10-27:
The Arduino core for the ESP32  sets `-Werror=all` when **File > Preferences > Compiler warnings** is set to "More" or "All", which upgrades warnings to errors:
https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L20-L21
The purpose is to force people to write quality code without warnings.

wjgeorge, 2019-01-10:
i agree with perl1234. This is probably bad code.

i get these failures on my Mac. I assume Xcode is more strict. Arduino version is 1.8.8

the date function as mentioned about needs a rewrite

this int function does not have a return value
     DS3231::begin

and several cases are missing in switches:
      WSop_close
      WSop_ping
      WSop_pong

and in src/sensors/ir/IRremote.cpp

there is bad code near if (irparams.blinkflag) {