hardkernel/ODROID-GO #11: Bad link for Arduino Setup on wiki.odroid.com

gojimmypi, 2019-06-11:
On this page: https://wiki.odroid.com/odroid_go/arduino/36_colorview

There's a link in the orange box at the top of the screen: "Make sure that you've followed the Arduino setup guide." - the link for Arduino Setup goes here, prompting for a password:

https://wiki.odroid.com/internal/playground/01_arduino_setup

It should probably go here: https://wiki.odroid.com/odroid_go/arduino/01_arduino_setup

awesometic, 2019-06-12:
Hi.

Thanks for your report.
I just corrected those links. It looks I have to edit the other examples too. :)

joshua-yang, 2019-06-12:
Thanks for your report, I just correct the bad links on the Wiki.
I think I have to correct on the other examples. :)

closed 2019-06-12 by joshua-yang