hardkernel/ODROID-GO #24: fatal error: hwcrypto/sha.h: No such file or directory

yoursunny, 2021-10-02:
* arduino-esp32 version: [2.0.0](https://github.com/espressif/arduino-esp32/releases/tag/2.0.0)
* ODROID-GO version: 4a496e337d16bca4ddedbeca3486d7b60662d017

When compiling the Hello\_World example with arduino-esp32 2.0.0, compilation fails with the following error:

```text
C:\Users\sunny\Documents\Arduino\libraries\ODROID-GO\src\web\WebSockets.cpp:42:10: fatal error: hwcrypto/sha.h: No such file or directory
 #include <hwcrypto/sha.h>
exit status 1
Error compiling for board ODROID ESP32.
```

This is because arduino-esp32 deleted the `hwcrypto/sha.h` header file.

dl9sec, 2021-11-01:
The workaround seems to be to replace
`#include <hwcrypto/sha.h>`
in
`...\Arduino\libraries\ODROID-GO\src\web\WebSockets.cpp (line42)`
by
`#include <esp32/sha.h>`

(see https://github.com/espressif/esp-idf/issues/3245)

Regards, Thorsten

starlight-w, 2022-03-03:
It not work.

starlight-w, 2022-03-03:
> The workaround seems to be to replace `#include <hwcrypto/sha.h>` in `...\Arduino\libraries\ODROID-GO\src\web\WebSockets.cpp (line42)` by `#include <esp32/sha.h>`
> 
> (see [espressif/esp-idf#3245](https://github.com/espressif/esp-idf/issues/3245))
> 
> Regards, Thorsten

It not work.

yoursunny, 2022-03-04:
This library currently has lots of non-essentials that are incompatible with arduino-esp32 2.0.
If you just want to support hardware on the ODROID-GO device itself, but not what you might connect via the I2C port, you should only keep the following files/folders and delete the rest:

* `library.properties`
* `src/odroid_go.*`
* `src/Fonts`
* `src/utility`

These files can compile correctly with arduino-esp32 2.0.

In CI environment, you can refer to the two lines linked here:
https://github.com/yoursunny/esp32cam/blob/b20ffa9ad2052e22355d4b7096166f0d5bc6d6f9/.github/workflows/build.yml#L42-L43

MarkKuhmann, 2022-05-20:
> The workaround seems to be to replace `#include <hwcrypto/sha.h>` in `...\Arduino\libraries\ODROID-GO\src\web\WebSockets.cpp (line42)` by `#include <esp32/sha.h>`
> 
> (see [espressif/esp-idf#3245](https://github.com/espressif/esp-idf/issues/3245))
> 
> Regards, Thorsten

worked for me when compiling the esp32_ble files of the Ardumower project in PlatformIO for the ESP32 dev kit c v4 (may be a AZdelivery specific name),

many thanks

Sarah-C, 2022-09-03:
I'm using Arduino 1.8.15 and I need it to compile because I've erased the partitions in the flash by accident so flashing the firmware doesn't work.
I'm hoping flashing the Arduino stuff will reproduce the partitions...

After making the fix :

> The workaround seems to be to replace #include <hwcrypto/sha.h> in ...\Arduino\libraries\ODROID-GO\src\web\WebSockets.cpp (line42) by #include <esp32/sha.h>

I then got:

In file included from C:\Users\sarah\Documents\Arduino\libraries\ODROID-GO-master\src\sensors\esp32-hal-gpio.c:20:
C:\Users\sarah\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/sdk/esp32/include/esp_hw_support/include/esp_intr.h:8:2: warning: #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead [-Wcpp]
 #warning esp_intr.h is deprecated, please include esp_intr_alloc.h instead
  ^~~~~~~
C:\Users\sarah\Documents\Arduino\libraries\ODROID-GO-master\src\sensors\esp32-hal-gpio.c: In function '__pinMode':
C:\Users\sarah\Documents\Arduino\libraries\ODROID-GO-master\src\sensors\esp32-hal-gpio.c:84:24: error: 'rtc_gpio_desc' undeclared (first use in this function); did you mean 'rtc_io_desc'?
     uint32_t rtc_reg = rtc_gpio_desc[pin].reg;
                        ^~~~~~~~~~~~~
                        rtc_io_desc

I found this to solve it but I'm not using the Espressif compiler!:

> Just in case this help someone the rtc gpio in menuconfig is in:
> Component Config -> Driver configurations -> RTCIO configuration

Hm I don't have that.
Also:

> The repository you are referring to seems to be written for IDF v4.0 or later. This is the reason why you are getting the errors: the code uses the new event handling mechanism introduced in v4.0 and not present in the earlier versions.
> You may try updating your IDF to v4.0, which was released today.
> 

Hm I don't have that either.
I opened esp32-hal-gpio.c, and replaced the function below with a return statement:

```
extern void IRAM_ATTR __pinMode(uint8_t pin, uint8_t mode)
{
return;
}

```

Then I got lots of errors:

In file included from C:\Users\sarah\Documents\Arduino\libraries\ODROID-GO-master\src\sensors\Adafruit_BME280.cpp:18:
C:\Users\sarah\Documents\Arduino\libraries\ODROID-GO-master\src/sensors/Wire.h:44:5: error: 'i2c_t' does not name a type; did you mean 'ino_t'?
     i2c_t * i2c;
     ^~~~~
     ino_t

It's a humidity sensor library - I wont be using that! So I deleted Adafruit_BME280.cpp, .h and .md

Adafruit_TCS34725.cpp was next..... deleted it and the .h file.

(I think by now, I should be updating some OTHER group of files somewhere.... or maybe these are just all out of date libraries, in which case as I'm not using them, it's easier to remove them than find the most recent and use it - I wish Odroid would keep their libraries up to date.... I guess 4 years is long enough to support it?).

RTClib.cpp was next with errors, so I deleted it and it's .h file. Real time clock? pah! I just need the "Hello world" to compile!

I'm nearing the end of the sensor list of files now. Yay!

Rtc_Pcf8563.cpp....... gone!

Ah! I see Wire.cpp isn't compiling, probably the cause of all those others if I'd bothered to read the full error.

Perhaps I need to look at the files Wire.cpp is using, maybe I need to copy a newer version file from some other esp32 folder?

For now I delete the sensors folder and recompile!

So - in total I deleted the content of the interupt setPin function, and removed the Sensor folder.

The "Hello World!" example now compiles! Yay!

I tried flashing the Odroid with the ESP32 download flash tool, and it...... doesn't boot anymore.

This whole attempt didn't work!  =(