hardkernel/ODROID-GO #25: Obsolete product?  How about releasing the design files, then?

runnerpack, 2022-12-22:
Hi, friends! 👋🏻
I have enjoyed using this excellent product for both fun and education, but I feel it could be so much more than it is.
If I had the PCB design files, I could use the compact-but-comfortable form-factor with the latest and greatest hardware inside!
If I had the CAD files for the case, I could print new backs to allow for more interface ports, bigger batteries, etc.
What do you say?  Good idea, right? 😉
Thank you for your time and consideration!

32teeth, 2022-12-23:
If you are interested, i open source my hardware (it fits a gameboy pocket case perfectly)
[GitHub Repo](https://github.com/retro-esp32/RetroESP32-Hardware)

|  [image]    |   [image]     |
| ---- | ------ |
| Top  | Bottom |

runnerpack, 2023-01-03:
Wow, @32teeth , that's really cool!